OpenSprinkler/OpenSprinkler-Hardware #1: Able to include a BOM?

dwjp, 2017-10-07:
Is it possible to include a BOM with component manufacturer and part numbers (and ideally distributor, e.g. Digikey, part number) for V3? I can probably figure out most details from schematic, but it'd be great to have something concrete.

adrian78666, 2017-10-07:
I second this request.

giig1967g, 2018-04-02:
i would like it too

laurentopia, 2018-08-21:
Have you folks figured out the BOM?

rubengr, 2019-12-09:
Is not public the BOM?

laurentopia, 2019-12-14:
here is the BOM so far, I gave up
https://docs.google.com/spreadsheets/d/11xkz92zQxZZgTctV5IspjD9BH1N4TvmNcIJ9ATqKawM/edit?usp=sharing

Egyras, 2020-01-17:
who can share AC ver. gerber file ?

Egyras, 2020-01-30:
> here is the BOM so far, I gave up
> https://docs.google.com/spreadsheets/d/11xkz92zQxZZgTctV5IspjD9BH1N4TvmNcIJ9ATqKawM/edit?usp=sharing

Hi, who has AC ver. BOM file ?

rayshobby, 2024-05-15:
1) We use a Chinese factory to manufacture the boards and they source the parts for us locally from China. So we don't have parts numbers in Digikey or Mouser because they don't buy parts from them.
2) The schematic has part values. Almost all parts are very common resistors, capacitors, diodes, transistors. A few ICs are not as common (like XL2001 switching regulator), but those parts are also clearly defined in the schematic.