OpenSprinkler/OpenSprinkler-Hardware #2: It would be great to have photos of how things look.

laurentopia, 2018-08-04:
I will build this project as a way to learn how to build circuits.
I can create a BOM in eagle but some items are not recognized, the switches and the plugs for example, that would be great to have your BOM in .csv format that mouser recognizes. Or maybe you use another store for your SMD? Also it would be great to have photos of how the parts fit.

laurentopia, 2018-08-04:
I tried to auto BOM->excell cleanup->mouser and got this list ;)
https://www.mouser.com/bom/edit?bomId=f5f01cde-92ca-4a33-b411-7b9ec5b9296c

rayshobby, 2020-06-19:
this cart link is your private link and is not publicly accessible.

closed 2020-06-19 by rayshobby

laurentopia, 2020-06-19:
this works:
https://www.mouser.com/ProjectManager/ProjectDetail.aspx?AccessID=b38ef469f6