OpenSprinkler/OpenSprinkler-Hardware #6: Schematics in PDF?

ilium007, 2020-10-15:
Hi - I don't have or use Eagle. Would it be possible to see the schematics in PDF format? Cheers.

ilium007, 2021-06-22:
Came here looking for PDF's schematics again and realised I already raised an issue for them! Any chance of PDF schematics?

lbussy, 2021-06-22:
Eagle has a [free version](https://www.autodesk.com/products/eagle/free-download) which I use for all of my designs.  You can [download it](https://www.autodesk.com/products/eagle/free-download), install, and [print the schematic to PDF](https://www.digitaltrends.com/computing/print-pdf-windows/) (part of Windows) in about as much time as it took me to respond to this.

ilium007, 2021-06-29:
I realise this but the schematic had no parts values, I just thought it might be easier to have a PDF included in the source files as is done on so many other projects I follow. I use KiCAD not Eagle so I don't have it installed locally.

rob040, 2022-07-19:
I second the request.
Or better, add PNG of schematic and pcb's and add full BOM, especially when schematic don't come with component values.

The @[lbussy](https://github.com/lbussy) suggestion to download Eagle yourself is not as straightforward as it sounds. First, you have to register at Autodesk and part with your privacy to be able to use it. Next, there is issue #6, where it is reported that files can't be opened.

Therefore, the request to the team to make schematics, PCB prints and bom also available in general readable formats.

rayshobby, 2024-05-15:
All current versions of board designs have PNG exports included in the folders.

closed 2024-05-15 by rayshobby