OpenSprinkler/OpenSprinkler-Hardware #8: Under which License is this repo published?

StohanzlMart, 2021-06-11:


StohanzlMart, 2021-06-12:
I'd really appreciate a response before Monday where I present OS as the hip tool it is and can be.

StohanzlMart, 2021-06-12:
If you can't answer the Issue-Question, then please answer: "Why isn't a License provided?"

lbussy, 2021-06-12:
From the GitHub page on [Licensing a repository](https://docs.github.com/en/github/creating-cloning-and-archiving-repositories/creating-a-repository-on-github/licensing-a-repository#choosing-the-right-license):

> [W]ithout a license, the default copyright laws apply, meaning that you retain all rights to your source code and no one may reproduce, distribute, or create derivative works from your work.

However, there's nothing stopping you from creating a PR adding say the MIT license.  Alternatively, you could email  support@opensprinkler.com as the organization page indicates.  If you really need the answer, asking before a weekend seems like it would have been a better choice.

Some of the other repositories in the organization are licensed under [AGPL](https://github.com/OpenSprinkler/OpenSprinkler-App/blob/master/LICENSE) and [GPL](https://github.com/OpenSprinkler/OpenSprinkler-Firmware/blob/master/LICENSE.txt).

StohanzlMart, 2021-06-12:
Thank you very much @lbussy for clarifying and the word of advise, much appreciated!

closed 2025-05-12 by rayshobby