OpenSprinkler/OpenSprinkler-Hardware #10: Enhancement suggestion for OSPi152.stl

daz10000, 2022-10-31:
Small suggestion but making a perforation or opening for the ethernet port would be handy - I guess people just deploy these things on wifi, but if you want to do a wired connection you need a drill :) - I should have thought of this before printing and just made a hole - my bad.
[image]

Sputnick85, 2024-01-03:
It would be great to have access to the USB Ports as well :)

rayshobby, 2024-09-07:
I will have to leave this for the user to modify the 3D model themselves. Different RPis may have different ports and positions (for example, the USB and Ethernet jack on RPi 4 and RPi 5 are not the same). Since I don't own all versions of RPis, I won't be able to create one variant for each RPi.

closed 2024-09-07 by rayshobby