OpenSprinkler/OpenSprinkler-Hardware #11: Burned Transistor Reeplacement

PipeDeveloper, 2023-01-14:
Hello @rayshobby 

Yesterday one of the station of my OpenSprinkler Pi stopped working suddenly.
After check the wiring and the valve then i move to check the OSPi board and i see a burned component on the board. (evident)

[image]

[image]

After some measuring 24v (AC) and a average consumtion of 0.23 - 0.25 A (per valve individually) i have no idea what could happen.
I dont know if the time of use is important on that case but specifically most of valves works about 20minutes per day except that station that usuay works for 3hours per day.

I would like to reeplace that component, but i could not find any information bout the part number, model, etc...
I apreciate your help in provide that information and maybe an alternative (that supports more current) to extend the life of the board.

Thank you in advance!

Henrik99, 2023-05-16:
By some google search I believe the component is the ST Z0103MN 6AA4. You can see the ST logo on the Triac, ST uses the Z3M as the marking for it (see page 14 https://www.mouser.de/datasheet/2/389/z01-1594011.pdf) I could be wrong tho.

rayshobby, 2024-05-15:
The triac is of type Z0103MN. There are several subtypes, but Z0103MN 5AA4 are quite common and widely available. If your hardware order was within 1 year, it's covered under warranty and we can replace it free of charge. For support questions please go to support.openspirnkler.com and file a support ticket there.

closed 2024-05-15 by rayshobby