OpenSprinkler/OpenSprinkler-Hardware #13: Updated Eagle Files or migrate to KiCad

AUPoolUser, 2025-04-06:
Hi 

it appears that the eagle files for the hardware are in the old binary format which are no longer supported but autodesk. Nor does KiCad appear to be able to import them into its latest version.

Any chance that the files can be updated to the latest format now that Eagle is practically dead unless you buy the full Fusion subscription. Alternatively kiCad would be great.

Or can you point me to an eagle download that is compatible? 

Thanks + Regards

rayshobby, 2025-05-12:
Which specific hardware design files is this referring to? Most of the recent hardware designs are in ASCII/human-readable formats.

rayshobby, 2025-05-14:
https://eagle-updates.circuits.io/downloads/latest.html
Version 9.6.2 does not require a paid license. It has limitations but the OpenSprinkler circuits are all below the limitations so can open just fine.

closed 2025-05-14 by rayshobby